MilesEngineering/MsgTools #14: Make project generator CLI tool

mgazic, 2017-11-16:
Put instructions https://github.com/MilesEngineering/MsgTools/wiki/Starting-a-New-Project when complete.

The tool might be "msgproj", and it might have these options:

- "language"
    takes name of new parser plugin, new file extension, name of existing plugin to base new plugin on
    makes a directory with a setup.py, so plugin can register itself
- "messages"
    makes a 'messages' dir, with Network subscription/connect/etc, and also a HelloWorld.yaml with ID 1
    makes a makefile with all *known* languages included
- "serverplugin"
     takes name of a new server plugin, makes a new plugin
        source has blank for user to enter their own tx code
        user needs to figure out what signal they can connect to, to read data and then emit the rx signal (or call rx function)
    with a setup.py, to package it (with plugin entry point)
    with a makefile to bundle, testupload, upload the plugin to pypi
-  "devel"
    does a git clone?
    does a "svn export <repo>/trunk/<folder>" from github (which allows svn access), to get *just* files without repo?

mosminerCP, 2018-06-06:
There is now a command line tool for generating web apps.  See #72.  I wrote this in the context of a pip package, which isn't the standard MsgTools think.  Perhaps it can serve as a prototype for a more robust tool to create apps for other languages as well?

mgazic, 2018-10-15:
Instead of making a tool to do it, I put up an example git repo here:
https://github.com/MilesEngineering/MsgToolsExample

All that's in that repo is some example YAML files, and a Makefile to run the code generator.

closed 2018-10-15 by mgazic