MilesEngineering/MsgTools #29: MsgServer: ['bytearray' object has no attribute 'raw']

mosminerCP, 2017-12-28:
The full error message displayed in Msg Server is "Exception in server.py while sending to client TCP Client ::1: ['bytearray' object has no attribute 'raw']"

Easy to reproduce - build the test messages, fire up two msgscopes, and try sending a message.  I just uninstalled and reinstalled  msgtools, and pulled a new copy of MsgTools before rebuilding my test messages vis a vis make test at the root of MsgTools.

It appears MsgServer is receiving the entire message, however when it goes to send I see the above error message pop out on the MsgServer console.

Tracking this down a bit it looks like the TCPServer is reading a byte array off the incoming socket vis a vis a QtDataStream, then initializing a new header with the byte array.  Subsequently when the TCP connection's sendMsg function is invoked it calls rawBuffer() and invokes raw on the returned byte array.

What's awkward is headers (messages in general) initialized without a buffer, create their own buffer via ctypes.create_string_buffer.  Which is why raw is required.  I suspect this is also why the copy trick is required in rawBuffer (wrapping the c_char in a dictionary) to avoid creating duplicates.  You can pass byte arrays around by reference without any copying going on.

I suggest we change the ctypes call to create a bytearray, and assume our underlying buffer is always a bytes or bytearray object (which has ripple effect on the accessor methods etc).  That will let us eliminate the raw calls, drop all the added overhead of managing a c_char object and integrate more cleanly with Python IO functions in general. 

In any case, here's a hack fix for TCPClientConnection (I didn't check Websockets)

def sendMsg(self, msg):
        buf = msg.rawBuffer()
        if hasattr(buf, 'raw'):
            buf = buf.raw
       self.tcpSocket.write(buf)

mgazic, 2017-12-28:
This sounds very familiar, but I thought I fixed it just a few weeks ago by making sure everything *did* have a c_types buffer, and calling "raw" on it would work.  Oddly, I do not see any errors on linux, which makes me wonder if there is a version problem (although it doesn't seem there could be).

mgazic, 2017-12-28:
Is it possible you had old generated Python code laying around?  Try a "make clean", then "make test", and run server/scopes again, please.

mgazic, 2017-12-29:
Can you take a look at the constructor for your NetworkHeader, and compare it to what you see in [msgtools/parser/python/HeaderTemplate.py](https://github.com/MilesEngineering/MsgTools/blob/master/msgtools/parser/python/HeaderTemplate.py)?  The latest version only directly uses the passed in buffer if accessing a ".raw" on it succeeds, otherwise it catches the exception and does a create_string_buffer() and copies into that.  That change was made 23 days ago, and it's in both the github repo and several of the last releases on pypi.

mosminer, 2017-12-29:
I think anything is possible.  So I backed the truck up and did the
following

1. git pull in my MsgTools repo

From https://github.com/MilesEngineering/MsgTools
   3a69311..88b30b2  master     -> origin/master
Updating 3a69311..88b30b2
Fast-forward
 msgtools/parser/java/language.py | 34 +++++++++++++++++-----------------
 msgtools/parser/test/TestJava.py | 10 +++++-----
 2 files changed, 22 insertions(+), 22 deletions(-)

2.  pip uninstall msgtools
3. In Msg Tools... virtualenv -p python3 ve
4. ve/bin/activate.sh
5. pip install msgtools
6. pip install PyQt5
7. pip install pyqtgraph
8. make test
9. which msgserver to confirm my virtualenv etc.
10. msgserver &
11. msgscope &
12. msgscope &
13. Send TestCase1 message

No error!  Looks good now!  Best guess - I was somehow behind or had old
revs of tools or something.

closed 2017-12-30 by mgazic