MilesEngineering/MsgTools #39: MsgBandwidthtestEcho randomly drops connection

mosminerCP, 2018-01-24:
I've seen it drop the connection just sitting idle (bandwidth tester not sending) and during streaming.  I've also seen random hangs where the UI goes non-responsive.  I isolated it by leaving my setup in place (BandwidthTester->AndroidServer->Echo) and replacing the echo app with a bare bones Python script that literally just reads the bytes off a socket and writes them back out.  No header parsing or anything.  BandwidthTestEcho usually lasts 5-10 minutes tops.  The simple loopback ran all night and it's still going.  Here's the code for the simple echo if you want to try and replicate

```
import socket
import sys
import select

ADDR = sys.argv[1]
PORT = int(sys.argv[2])

print("Trying to connect to", ADDR, ':', PORT)
with socket.socket(socket.AF_INET, socket.SOCK_STREAM) as s:
    s.connect((ADDR,PORT))
    s.setblocking(0)
    print("Connected to", ADDR, ':', PORT)

    out_buf = b''

    while True:
        # only wait to read when the corresponding outgoing buffer isn't full
        # only wait to write if we have things to write
        ret = select.select([s],
                            [s]      if len(out_buf)>0 else [],
                            [],
                            10.0)
        if len(ret[2])>0:
            raise StandardError("had exception on socket")

        if s in ret[0]:
            out_buf += s.recv(4096)
            # print('Got', len(out_buf), 'bytes')

        if len(out_buf)>0 and s in ret[1]:
            sent = s.send(out_buf)
            out_buf = out_buf[sent:]            
            # print('Sent', sent, 'bytes')
```

Miles-Gazic-CardinalPeak, 2018-01-24:
I haven't been able to reproduce the error so far.  I tested using my home wifi, with tester, echo, server (either msgserver, or AndroidServer) on the folllowing OSes:
1) linux, linux, linux
2) win, win, win
3) win, win, linux
4) linux, linux, win
5) linux, linux, Android
I have not seen a dropped connection yet.  I am using AndroidServer built a few days ago, I have not updated to the latest AndroidServer today.

Miles-Gazic-CardinalPeak, 2018-01-24:
I let it ran for a while with lin/lin/Android, and then when I closed bandwidthtester, AndroidServer immediately showed *both* clients as 'Disconnected', and echo also immediately showed that it had been disconnected.  The zombies hung around in AndroidServer, and I was unable to get new clients to fully connect to AndroidServer.  Tester and echo showed they were connected, but they didn't show up in AndroidServer's Connections list, and although they could send data, they could not receive any.

mgazic, 2018-03-02:
Cannot reproduce the bug, closing this.  My hope is that it was somehow related to network comms with AndroidServer, and that with the latest AndroidServer it is not an issue.

closed 2018-03-02 by mgazic