MilesEngineering/MsgTools #64: Support an environment variable to specify generated code location

Matt-Osminer-CardinalPeak, 2018-03-08:
It would be super nice to be able to run msglumberjack and other tools from anywhere and not require the generated code to be at or higher level.

How about an environment variable you can set to force the code location?  Possibly even a command line option.

mgazic, 2018-03-08:
I think I like command-line option more than environment variable.  The bad thing about using an environment variable is that if you have multiple repos (or multiple branches of the same repo) on your PC, it obfuscates which version of generated code is being used.  If someone set the environment variable once they'd forget about it, and even to set it once some people would do it in the terminal and not close the terminal for a month, others would have to figure out if it goes in .bashrc or .profile and have to figure out if they need to restart the terminal or manually source the init script.

Right now the runtime Python code checks for generated code in two locations:
1) Relative to MsgTools source directory.  If you're running from github clone this is sometimes useful, but if you've installed MsgTools this won't be useful.  'make test' ends up creating an objdir and putting generated code for test messages in a place that the tool will find them, which is only a good thing if you realize that's happening.  It may be more likely that people will run unit tests and then get confused about why their normal messages aren't seen, and instead these strange fictional ones are used.  Also, when running 'make test', it creates an obj dir one level higher than the github clone dir, which is a legacy feature from having a Makefile to invoke the code generator inside the MsgTools repo, but wanting to run it on messages in a dir outside the MsgTools repo, and put generated code in a dir outside the MsgTools repo.  Not sure if either of these make sense anymore, it might be better to have projects that use MsgTools just write their own Makefile (which is what BMAP_SpecAndTools does).  And have 'make test' create an objdir inside MsgTools, not in the parent directory of it.
2) Relative to current working directory.

If I add the command-line option to specify msgdir, I may remove #1 above.  I will probably also have apps error out if they don't have a valid msgdir.  Right now the client apps start up without exiting but don't work very well, it'd be better if they quit with an explicit error.

mgazic, 2018-03-08:
--msgdir= option is now checked in to github, and also added to release 0.28.32

closed 2018-03-08 by mgazic