MilesEngineering/MsgTools #69: MsgServer Logging Odds and Ends

mosminerCP, 2018-05-14:
Looks like MsgServer is not filtering LogStatus, Note, and ClearLogs and is forwarding them on.  I've got MsgScope hooked up to a MsgServer and am just sending log msgs.  Looks like Start/Stop Log is being filtered, as we QueryLog.

Also looks like MsgServer is not handling ClearLogs as all my  logs are still on disk.

mosminerCP, 2018-06-06:
Did some more thorough testing on this:
* Log Start, Stop, and Query cause a status message to be sent and aren't forwarded.  This is ok.
* ClearLogs is not handled and is passed through to other clients
* Note if passed through to other clients.  Probably not the behavior we want.  We just want to log the note and drop it.

mgazic, 2018-06-06:
I'm hesitant to want to implement ClearLogs in msgserver.  Right now the user can save logs anywhere they want, with any file extension they want.  I'd hate to inadvertently delete files that shouldn't be deleted.  Also, on a PC there's tons of easy ways for the user to delete files if they want to (vs. android/iOS, where ClearLogs is much more necessary).

mosminerCP, 2018-06-06:
I can buy that argument.  I'll just silently eat ClearLogs.

closed 2018-06-06 by mosminerCP