MilesEngineering/MsgTools #70: Parallel build for code generator

mgazic, 2018-05-17:
Apparently there's a way to do I/O operations in parallel in python:

> For a nice speed up with IO operations consider using pythons multiprocessing Pool for dramatic speedup.  For example the following code (processing 200 yaml files with references) went from taking ~15 seconds to 4.5 seconds when run in parallel.
```
from multiprocessing import Pool
# Parallel method
with Pool(5) as mp:
```

We could look at having a single invocation of msgparser use this, to more quickly process a large set of .yaml files.

mgazic, 2018-05-17:
Care should be taken that any output messages, especially related to errors in parsing YAML, are still in order and make sense.

jkominek, 2018-05-17:
the `threading` module should be sufficient for parallel IO, as python IO operations all release the interpreter lock.