MilesEngineering/MsgTools #75: Multilog crash when clicking tag button

mosminerCP, 2018-05-31:
In ProcessMsgs there is an undefined hdr message that crashes the app when you hit the 'ButLabel' button.

This is the stack trace off the argparse branch @commit 794273d4ea0ca60a089a52ec86155146529cab1a

This is my command line:
 msgmultilog --fields FieldOne FieldTwo --buttons hotkey:X,tag:tagit,label:ButLabel --show BandwidthTest --plot BandwidthTest[SequenceNumber] --send BandwidthTest &

Traceback (most recent call last):
  File "...msgtools/multilog/multilog.py", line 214, in ProcessMessage
    self.file.write(hdr.rawBuffer())
NameError: name 'hdr' is not defined

mosminerCP, 2018-05-31:
If I checkout master (commit 9aa9a85c922677feddaff9a455aa645045847b6a) I see a similar issue.

Here's my command line - 

msgmultilog --field FieldOne --field FieldTwo --button hotkey:X,tag:tagit,label:ButLabel --show BandwidthTest --plot BandwidthTest[SequenceNumber] --send BandwidthTest

To reproduce:
1. Launch msg server and msgscope
2. Launch msgmultilog
3. Click the ButLabel 
4. Send a BandwidthTest message from MessageScope

Error output - 
File "..msgtools/multilog/multilog.py", line 194, in ProcessMessage
    self.file.write(hdr.rawBuffer())
NameError: name 'hdr' is not defined

mgazic, 2018-06-06:
Fixed by using msg.hdr instead of hdr.

closed 2018-06-06 by mgazic