MilesEngineering/MsgTools #76: Lumberjack socket support

mosminerCP, 2018-06-01:
Right now Lumberjack only supports a file input, however it is built on the core App framework which supports IP connections etc.  

We should augment Lumberjack to support connection to a server via TCP or Websocket and process directly?  With the current core App support that looks like a fairly simple bump up.  The argparse work has already laid the command line arguments down as well.

Miles-Gazic-CardinalPeak, 2018-06-01:
Previous versions of lumberjack did work with sockets, file, or serial I/O.  It's been years since anyone used that though, so it is undoubtedly broken.  I think the fact that people haven't clamored to use lumberjack in those other ways probably means most people don't care, and the value of making it work is low.

By the same token, at some point other Python apps did work with files, but that also hasn't been used for years.  For example, it used to be possible to open msgscope/msginspector with a file as input, and they'd display/plot all the messages from that file.

Alternatively to using lumberjack on a socket, we could also consider making msgserver log to multiple CSV files (right now only binary and JSON are supported), or make a msgserver plugin to support logging to CSV files.

mosminerCP, 2018-06-01:
Well, as I was converting everything over to use argparse for command line handling I did find a few options that didn't appear to work.  I fixed noisemaker (wasn't processing the list of messages correctly), but I hardly did an exhaustive test on everything, nor did I deeply familiarize myself with app.py to understand all the functionality and how it works.  

I think there is room to further improve my first cut at argparse conversion as well.  Some base app.py options don't make sense with some tools, and as you say I'm not clear every option works with every app.  I'll write up an issue to to a pass on all tools and options and clean things up just so it's a known thing.

mosminerCP, 2018-06-05:
The current app.py architecture, which handles all the core messaging, is built on Qt Sockets using signals.  This approach requires you to execute a Qt event loop in order for signals to be processed.  Alternately you can use the waitFor* APIs and implement your own worker thread.

File processing for lumberjack was (and still is) done via straight up Python files.

For now I did a little bit of a hack and if we're reading from a file, then I use the old MessageLoop/non-Qt method of processing.  If we're using sockets I execute the QtApplication event loop to pump signals.

I'll write an enhancement issue to refactor app.py a little bit to line up file and socket IO into a more common model. Meantime lumberjack will now consume socket traffic as well as process files.

closed 2018-06-05 by mosminerCP