MilesEngineering/MsgTools #82: Refactor app.py to better support socket and file sources

mosminerCP, 2018-06-05:
See #76 closing comments for more details.  The idea here is align the file and socket source options to make processing of messages consistent across files and sockets.  Right now we have a mix of Qt sockets with signals, and Python files.  Each source has different threading and handling models.

While I see some advantages going with Qt, I'd be inclined to use vanilla Python sockets and files so Qt is not a requirement for every application.  

During the refactor it might be fun to add the possibility to specify a list of files that could be chained together.  For instance, if you want to process multiple log files in one run.

Also with this, I would consider building the msg command line option in as a core feature, or move it out of the base class and into specific apps.  Right now msg is only supported by inspector and multi-log.

mgazic, 2018-06-06:
A long time back msgtools supported both qtsocket and socket.  The socket version (at least the way I wrote it!) was ugly compared to qtsocket, and I never ended up writing non-Qt apps.  If there's an easy way to use Python sockets that doesn't require anything funky like creating an extra thread, and easily integrates with Qt signals/slots, I'd definitely be open to it.  In my mind signals/slots with Qt works great for event-driven apps (whether GUI or CLI), and it's only for procedural apps that want to sleep that I wouldn't use Qt (and that's what 'console' is for).

mgazic, 2018-10-15:
I think one feature that would be pretty cool that we're missing is the ability to 'playback' a log file.  As long as the message header has time stamps, that should be easy to do with decent fidelity of inter-message timing.  Not sure if it should be a feature of msgserver, a plugin for msgserver, or a new client app (msgplayback?).  Also dunno how much of a UI it needs: play/pause, seek forward/backward, jump to a time?  Or just run all the way through from start to finish unless killed?  Apps like msginspector and plotting could get easily confused if time jumps around.

I'm fine with keeping this issue opened just as a note of the future work and cleanup to take place.  It's unfortunate that lumberjack uses the file read feature of app.py.  Maybe that should be moved to a different baseclass, instead of cluttering up app.py with a feature that no other app besides lumberjack is ever likely to use?  App.MessageLoop is pretty much all that's used for reading from a file, and the rest of App is used with qsockets.  Lumberjack is the only thing that uses MessageLoop, and every other app always uses qsockets.

mgazic, 2023-01-15:
[In September I created a new class MessageFileReader](https://github.com/MilesEngineering/MsgTools/commit/acc8d36ebd0f91c4403737ea810cfb06f8a7e9c6) and modified Lumberjack to use it.  It's optimized for reading from files, and separate from the msgtools.lib.gui.Gui which allowed simplifying a lot of code.

closed 2023-01-15 by mgazic