MilesEngineering/MsgTools #105: Python2 template support

mosminerCP, 2018-09-18:
Our current python templates are for Python3.  Create a Python2 template for those folks using older systems that only support Python2.

Maybe we have a python2 and python3 language type on the command line to support this?

mgazic, 2018-09-18:
Why bother supporting Python2?  Python3 has been out for 10 years.

mosminerCP, 2018-09-19:
Some older embedded Linux systems only support Python 2.  

Just to be sure we're on the same page; I'm talking about generating Python2 message code, not making all the MsgTools utilities run on Python2.

mosminerCP, 2018-09-24:
Turns out generated code (apart from the messaging lib) is Python2 friendly.  The main issue is the messaging library.  Which in and of itself is actually close to being Python2 compatible as well.  There are some class type (i.e. type()) calls that fail under Python2.   I could fix all this and make it work for both Python2 and Python3.

Ultimately Python2 is dead, and we really don't want to stifle forward progress by having to support and test both 2 and 3.  I wrote a [wiki page](https://github.com/MilesEngineering/MsgTools/wiki/Python2-Support) outlining details, but in a nutshell I have provided a command line option in msgparser that let's you spec your own messaging package, so you can write your own messaging module, stubbing it if you want, and still use the built-in templates.

If the templates ever become python2 incompatible, users can use the -t and -ht options to spec their own templates.  The extreme version of this would be writing a python2 language plugin if necessary.

I'll also put a python2 friendly messaging class in utils/python2 that will serve as a mostly working snapshot at this time.  Folks can copy this into their Python2 projects and use the -m msgparser option to map it in.  If somebody wants to maintain the python2 messaging lib, all the better.

mgazic, 2018-09-25:
Why not leave the name 'msgtools.lib.messaging' in the generated code, and force users who cannot use 'real' msgtools to write their own msgtools/lib/messaging.py file?  They can't make a pip package with that name of course, but we don't want them to anyway.

mosminerCP, 2018-09-25:
Nothing I've done precludes what you're saying, and I am basically forcing folks to write their own messaging file.  I'm just providing an easy and automated way to swap their own messaging file in and put it wherever they want.

I felt the ability to override the default messaging module might be helpful in other ways for both Python 2 and 3 and added the command line option.  It certainly helped me do quick and easy side by side testing with the built-in and hacky Python 2 messaging modules.

I kinda feel like we should split the messaging module out a bit.  Looks like there are a few distinct pieces of functionality in there.  

Message module loading and factory methods.  This is what the generated message code cares about is the message loading, and registration.  Feature idea - make a network based message loader.  Then you could generate and host message code on a central server and everyone could share the same version of code.  This could actually be a very cool way to deploy new versions of messages in general.

There are a variety of methods for format conversion (JSON, CSV, etc).  Handy, but not necessarily useful for every application.

There are network level utilities for header conversion and routing.  Again, not necessary for every app, but handy when you need them.

And finally a collection of reflection and utility methods that are handy for tools that want to use messages in a generic way.  I'd imagine most app developers will reference the message accessors directly, but if they need a lot of flexibility these functions are handy.

Splitting these into individual modules would make it easier to override, and pick and choose what you want.

mgazic, 2018-09-25:
> I kinda feel like we should split the messaging module out a bit.
> Looks like there are a few distinct pieces of functionality in there.
What do you think a better layout would be?  Right now there's "lib" in the path, which maybe isn't very Pythonic: everything under a package is part of that "library", and nobody calls it a "library" in Python either.  If anything the applications should be under a separate name ("apps"?) to distinguish from package code intended to be used by users of msgtools.

Anyway, under lib there's currently:
app.py
gui.py
messaging.py
msgplot.py
txtreewidget.py
unknownmsg.py

Would it be sufficient to just split out the JSON and CSV functionality into json.py and csv.py?  Maybe split HeaderTranslator and MsgRoute out to a new "header.py" (or other better name), too?

The "MsgRoute" isn't really for routing messages (because everything is broadcast), it's for making a unique tag (which is termed a "Route") using content from the header, to know if messages with the same ID are to/from different devices, so they get displayed separately in the UI.  We don't want the same message from two devices showing up as alternating points on a plot, for example.  MsgScope also uses this in it's receive pane so you can see if you've gotten the same message from two different devices, and show them as separate entities in the lower-right pane too.  MsgRoute tries to do it in a generic way without presuming anything about how the user has set up their header (assumes any header field that doesn't contribute to ID, and isn't length or time, has something to do with Route).

The FieldInfo/BitFieldInfo and Messaging.set/.get functions are pretty intrinsic to Python (and used in the auto-generated code), I think I'd prefer to leave them in messaging.py.

> Feature idea - make a network based message loader. Then you could generate and host
> message code on a central server and everyone could share the same version of code.
> This could actually be a very cool way to deploy new versions of messages in general.
I like this, and have done similar things in the past, but I'd like a really generic way to do it.  Previously I had the ICDs stored in the devices and tools would query them, cache them locally, all msgtools would look in the cache and parse the ICD at runtime.  That was sort of cool, but presumed way too much about how the system worked to ever belong in a generic msgtools package.  What I'd be more comfortable with is making it easier for users to manage multiple copies of their auto-generated code outside of msgtools, and then run msgtools apps for the appropriate version.  A part of that might be the new 'launcher' app I started (https://github.com/MilesEngineering/MsgTools/tree/launcher), which has buttons (soon to be icons) to launch all the GUI msgtools, from the same dir that the launcher started in.  That way if people are using GUI tools they don't have to open terminals, cd to the right directory, and type commands.  They'd instead invoke the launcher correctly once, and the launcher will start all the msgtools apps from the right directory so they use the right autogenerated code.

mosminerCP, 2018-09-25:
Yep I think this kind of split makes sense.  Not mandatory though.

mosminerCP, 2018-09-27:
Addressed by Pull Request #109.

closed 2018-09-27 by mosminerCP