MilesEngineering/MsgTools #107: Tests Failing

mosminerCP, 2018-09-19:
(ve) Matts-MBP:MsgTools mosminer$ make test
==> make /Users/mosminer/Projects/Bose/MsgTools/ThirdParty 
make[1]: Nothing to be done for `test'.
==> make /Users/mosminer/Projects/Bose/MsgTools/msgtools 
==> make /Users/mosminer/Projects/Bose/MsgTools/msgtools/parser 
/bin/sh: line 0: cd: ../../../messages: No such file or directory
cd test && make
python3 TestCpp.py
....
----------------------------------------------------------------------
Ran 4 tests in 0.039s

OK
python3 TestPython.py
....
----------------------------------------------------------------------
Ran 4 tests in 0.037s

OK
python3 TestJava.py
....
----------------------------------------------------------------------
Ran 4 tests in 0.041s

OK
====> Checking message validity
python3 ../check.py /Users/mosminer/Projects/Bose/obj/CodeGenerator/MsgDigest.txt messages

Error in messages/BandwidthTest.yaml!

Traceback (most recent call last):
  File "../check.py", line 144, in <module>
    main()
  File "../check.py", line 137, in main
    ProcessDir(outFile, msgDir, "")
  File "../check.py", line 36, in ProcessDir
    ProcessFile(filename, outFile, inputData, subdirComponent)
  File "../check.py", line 115, in ProcessFile
    PatchStructs(inputData)
NameError: name 'PatchStructs' is not defined
make[3]: *** [/Users/mosminer/Projects/Bose/obj/CodeGenerator/MsgDigest.txt] Error 1
make[2]: *** [test] Error 2
make[1]: *** [test] Error 2
make: *** [test] Error 2

mgazic, 2018-09-19:
Tests pass when I try here.  Are you using latest release, or code from github?  If github, is it possible the last pull didn't grab everything, or that there's a version of msgtools that's installed that's conflicting with the 'make develop' on the github version?  What do you get if you cd msgtools/parser/ ; grep Patch *.py?

mosminerCP, 2018-09-19:
Argh!  Yea, looks like had a pip install conflict.  Sorry about the noise!

closed 2018-09-19 by mosminerCP