MilesEngineering/MsgTools #111: Messaging Refactor

mosminerCP, 2018-09-27:
Refactor how generated code uses Messaging to try and remove the coupling so that messages can be used stand alone, and move to a lazy loading scheme to reduce memory load, and save CPU by not loading all messages and loading only what we need.  

Longer discussion at the end of #109 but the nutshell idea at present is:
- Generate a top level __init__.py that registers all the messages with Messaging
- If the user doesn't provide a messaging module just move on
- Use factory methods across all tools
- Factory methods load modules when needed
- Minor, but while we're considering it, look at splitting messaging into smaller more specific modules to make it easier to write your own

Concerns:
- Need to retain the decorators and Field Info functionality somehow
- Evaluate toCSV and similar methods; will these work correctly with a lazy loading scheme?

An alternate (and easier to implement) idea would be to support a "template light" for each language.

Need to more deeply evaluate options.

mgazic, 2018-09-28:
> If the user doesn't provide a messaging module just move on

I don't see the benefit of that.  The things we need to have at a minimum are decorations, FieldInfo class, and Messaging.hdr.  I know that existing python apps use the decorators very sparingly, if at all.  Nearly all the same info is contained in the FieldInfo structures in a cleaner way for reflection-based apps.  *However*, if we assume the future existence of more Python apps that use specific messages by name instead of reflection, then it's also reasonable to assume that those apps would make heavier use of the metadata in decorators.  I do not think it makes sense to try to get rid of both decorators and FieldInfo.  One or the other (but probably not both) would be very important to have in any given style of application.  I don't really see what the problem is of requiring a single file named "msgtools.lib.Messaging" to exist, either.  Agreed that the existing one has too many perhaps unrelated features, but splitting those into separate msgtools.lib.json, msgtools.lib.csv, msgtools.lib.header, seems like a good solution for that.

> Evaluate toCSV and similar methods; will these work correctly with a lazy loading scheme?

I assume you don't mean converting a message object to CSV (like the current implementation of Messaging.toCSV), but instead starting from an array of bytes (received over network, serial port, etc.) instead of a message object.  To do that now, the first step would be to call Messaging.MsgFactory(), and that's what you'd have to replace with a new version that does lazy loading.  If someone wants to use reflection, IMO the current implementation of loading all code on startup works fine.  There's no perceptible delay on startup due to loading all the Python.

If you do want lazy loading, the cleanest way I can think of doing it would be with a cache of ID->filename.  We'd keep all the functionality that exists now, but also at runtime rely on a cache file with a map of message ID->generated file name.  If the map doesn't exist it'd be created on app startup by loading all the generated files.  If the file does exist (from a previous invocation of the app), as long as it's newer than all the generated files listed in it, those files won't be opened on startup, but instead lazy-loaded as needed.  This would mean first invocation would be same speed as currently, and subsequent invocations could be faster.

> Use factory methods across all tools

I'm not sure what you mean by that.  Right now all received messages do go through a message factory to convert the byte array to a message object.  For creating new messages, I really like the way that the code currently does that with Messaging.messages.AudioManagement.Get, and I don't want to switch to Messaging.Factory("AudioManagement.Get") just for aesthetic reasons.  To make that work with lazy loading, you'd have to reimplement __getattr__() like this: https://stackoverflow.com/a/5517322

Regarding light weight embedded systems that don't need reflection and don't want to load any generated files that aren't used, I don't think we'd have to change Messaging much to acheive that.  Looking at the implementation of the Messaging class, there are *no* member variables.  There is a constructor that does the recursive load, but that could be changed to a static LoadAllMsgFiles() function.  There's a couple functions not marked as static that don't use 'self' and are only invoked by Messaging.fnname(), and those should be marked static.  The only function that does take a 'self' as param and is invoked as obj.fnname() is LoadDir, and that should be changed to static, too.

After that's done,  if someone didn't want to use reflection and didn't want to load all the Python code at startup, they could simply not call the LoadMsgFiles() function, and instead call some new function LoadHeaderFile(headerfilename) that just loads a header and sets Messaging.hdr.  All the auto-generated python code could still be imported manually one file at a time.  Any auto-generated python files they did manually import *would* call Messaging.Register, but all that will do is add that class's name and ID to a couple hash tables.  It's a good thing that Register is called, because that would let them use the existing MsgFactory() on received byte arrays for just the messages they care about.

> An alternate (and easier to implement) idea would be to support a "template light" for each language.

Can you elaborate on that?  I think this issue is Python specific, and shouldn't change how we do things in Matlab, Javascript, Swift, Java, Dart, etc.  There's no way we'll find one strategy that works for all languages.  Currently C is more "light" than anything we've talked about for a "light" Python.  I'm not going to implement a 'heavy' C with full reflection.  C++ does have reflection, but protected by #ifdef so it switches between "very light" and "sort of light".

mgazic, 2018-09-29:
I'll refactor Messaging.py this weekend to make all functions static, and change from a constructor to a LoadMessages(path, headerclassname) function, and add a LoadHeader(headerfilename) function.  Users will need to do one of either LoadMessages() or LoadHeader().  I'll also update msgtools/lib/app.py to use LoadMessages() instead of constructing a Messaging object.  That'd be safe to merge right away, and after that changes shouldn't need to be done to the Messaging API, just implementation.  After that, we can potentially do another branch to add lazy loading (perhaps via the caching scheme I outlined above).

mgazic, 2018-09-29:
I created a branch to clean up Messaging, and made all methods static, and got rid of constructor.  I unfortunately also accidentally pushed a change to do lazy-loading, too (should've been in another branch).  It's all in this branch:
https://github.com/MilesEngineering/MsgTools/compare/messaging_refactor

mgazic, 2018-09-29:
I timed lazy loading vs. loading all messages at startup, on my Dell XPS13 running Ubuntu 16.04 with stock Ubuntu python3:

with msgtools test msgs:

    Using lazy loading, 0.0001976490020751953 seconds
    Using lazy loading, 0.0002522468566894531 seconds
    Using lazy loading, 0.0001857280731201172 seconds
    Using lazy loading, 0.00018978118896484375 seconds

    Loading all msgs, 0.008506536483764648 seconds
    Loading all msgs, 0.008158445358276367 seconds
    Loading all msgs, 0.008475303649902344 seconds

with Bose BMAP messages

    Using lazy loading, 0.002207040786743164 seconds
    Using lazy loading, 0.002208709716796875 seconds
    Using lazy loading, 0.0022745132446289062 seconds
    Using lazy loading, 0.0034325122833251953 seconds

    Loading all msgs, 0.10916733741760254 seconds
    Loading all msgs, 0.11308884620666504 seconds
    Loading all msgs, 0.11041903495788574 seconds
    Loading all msgs, 0.11113834381103516 seconds

mgazic, 2018-09-29:
I figured out a regex to search the code for likely use of Messaging decorators.  The decorators are a property of an auto-generated SetXXX/GetXXX function, and start with a lower-case letter.  The function doesn't get invoked: it's just the name followed by .decoratorName.  So this find | xargs grep I think shows them all:

    find . -iname \*.py | xargs grep '[SG]et[a-Z]*\.[a-z]' --color

There's only 11 uses right now.  They all look good to me, and changing them to use FieldInfo would be awkward (which is why I did them with decorators to begin with).  Like I said earlier, apps that use reflection don't use decorators much, but new apps that *don't* use reflection are likely to use decorators more.

mgazic, 2018-10-19:
Things got shuffled quite a bit (features like csv/json moved out of Messaging, lazy load implemented, new Message class added in new message.py), and I think the goal of this has been achieved.

closed 2018-10-19 by mgazic