MilesEngineering/MsgTools #138: WebConsole fails because it can't find .js files

mosminerCP, 2019-01-25:
if you make the WebConsole project it generates the WebConsole app but the app doesn't work because ../MsgApp/js/msgtools.js is missing.

mosminerCP, 2019-01-25:
Two things - I created a webapp utility to generate web apps from a common library to avoid version mismatch.  When I did that I moved the missing file to msgtools/lib/js.  

Second I didn't convert WebConsole to use the new webapp utility.  So a whole bunch of stuff was missed.  I've hacked a new Makefile to use webapp and copy the message definitions in.

Should add a command line option to webapp to manage the message copy as an option in order to let you create standalone web apps.  I'll work on that in a bit.

mosminerCP, 2019-02-12:
@mgazic I've modified the webapp generator to copy message files in for WebConsole and make a full standalone web app.  My last  hurdle here is how to inject the message directory into the makefile.  For testing I generated the test messages at the root of MsgTools so the makefile has a hard coded reference to ../obj/CodeGenerator/... for now.

I think the correct end solution is to make the web app generator mirror messaging.py in terms of searching for generated Javascript code.  I can still include the command line specifier to override the search and let an end user pick, but otherwise we'll search up the directory tree for javascript just like the Python side.  What do you think?

mgazic, 2019-02-12:
I think there's two use cases that are worth considering:

1) For development, it's really nice to have only one copy of all files, and things refer to them with relative paths.  That way people can edit them and hit 'refresh', and they never get confused about needing to do other stuff to make their changes take effect.  As soon as copies are made, it's inevitable that the automation to make the copies will not quite be good enough (often because humans will circumvent it by editing copied files).
2) For deployment, we maybe want a bundle that has everything that needs to be deployed, and nothing else.  I'm actually ambivalent about this because IMO it'd be fine for a webserver to do a 'git clone' and run 'make javascript', and then it'll work just like the development environment does.

For me personally, I'd like to focus on making 'development' mode as simple as possible, and have a separate app or makefile rule to bundle stuff for deployment.  WebConsole did need a build-time script to generate a file with include statements for all generated javascript files, which is unfortunately complicated for development.  Other simpler apps though could be hand-written to include only the generated files they want, and during development I'd like it if no build-time script was needed (except for running the code generator itself), and no copies of files needed to be made.

I think I'm likely oversimplifying things, and my ideal scenario of 'no copies' might not be feasible.  I'm mostly stating it in writing as a goal, even if we can't achieve it.  The issue of where to get hand-written .js files from when using installed msgtools (instead of git cloned msgtools) is certainly a sticking point, and I don't know what the best solution for that is.

mosminerCP, 2019-02-13:
I like the goal, but web constraints make it tricky.

FWiW I'm still effectively generating the include file - I've just
abstracted it behind a list of message names.  The idea was that when the
web app if first created we build the full list of known messages, and then
the user can weed out anything they don't want to reduce load time if
desired.  Adding new messages manually (or rerunning make) after that of
course.  The abstraction was really more of a convenience as much as
allowing me to define a set of must have messages to support all the
various tools.

Here's the magic:
    // Initialize MsgTools itself...
    msgtools.setMsgDirectory('messages')
        .then(() => {
            // Load app specific messages
            msgtools.loadMessages(['Printf', 'TestCase1', 'TestCase4',
'TestCase3', 'TestCase2', 'BandwidthTest', 'TextCommand', 'TextResponse',
'Taxonomy.Canidae.Canidae', 'Taxonomy.Canidae.Canis',
'Taxonomy.Canidae.AFox', 'Taxonomy.Canidae.Wolf',
'Taxonomy.Canidae.Vulpes', 'Taxonomy.Canidae.Fox', 'Taxonomy.Canidae.Dog',
'headers.SerialHeader', 'headers.BluetoothHeader', 'headers.NetworkHeader',
'headers.TaxonomyHeader', 'headers.CANHeader', 'Network.QueryLog',
'Network.LogStatus', 'Network.Connect', 'Network.PrivateSubscriptionList',
'Network.History', 'Network.ClearLogs', 'Network.Note',
'Network.SubscriptionList', 'Network.StopLog', 'Network.StartLog',
'Network.MaskedSubscription'])
                .then(() => {

I just had a brainwave on a potentially better way to manage this, but hold
onto that.

Fundamentally it's a path problem.  A web server won't serve files higher
than the root.  So a user must place message code at or below root and give
the WebConsole app a path to it.  Since code can be generated anywhere, you
necessarily need to tell WebConsole's make process where the messages are,
OR we need to settle on a standard location.  Either way, onus is on the
developer to set things up correctly if they want to use the generated code
directly.

Options:

   1. Pass the location of the messages into the make command, or build in
   a search like the Python tools use to discover message code.  The
   dev/deployer needs to make sure they run their server at a high enough
   directory level to work.
   2. Settle on a standard location for message code in WebConsole and
   leave it to the dev/deployer to generate code to the right place.  Dev
   still needs to remember to run the makefile to update the message list (or
   edit the source) if they add new messages.  Server can be run at WebConsole
   or higher level as desired.
   3. Copy generated code into a sub folder so WebConsole can be hosted at
   the root.  As you observe - good for deployers, bad for developers.

In circumspect, I'm thinking #2 may be the best solution.

Back to my brainwave, it occurs to me we could use WebDav as a means of
dynamically discovering message code on servers that support it.  In this
case we would search the server for message files at runtime and skip the
need for the "include" file.  Many downsides to this:

   - Adds a fair amount of complexity.  Need to find a suitable WebDav
   client, integrate it, etc...
   - Not all servers are guaranteed to have WebDav enabled or supported
   (though most production servers at least support it)
   - Appears the built-in Python HTTP server doesn't support WebDav.
   Plenty of packages to add it, but something else for devs to manage.
   - May substantially increase app load time since we have to scan the
   server.
   - Need some way to distinguish message code from regular Javascript
   hosted by the server.  It would be bad form to blindly load all the
   Javascript from the server.  Can be mitigated by letting the app developer
   specify the messages folder to search, which gets back to developer
   responsibility on where to drop the messages and tell the app about it.

This doesn't necessarily solve the overall path problem, but the upside is
it would allow us to provide a stock WebConsole app that doesn't have any
make smarts.  You could just generate your code, run a WebDav capable
server at a high enough directory level for WebConsole and message code,
and load the app.  The only tricky bit is you need to force reloads if you
change message code - though I might be able to figure out a way around
that too.  Would work for local and production cases as needed.

mgazic, 2019-02-24:
I fixed WebConsole in a branch here: https://github.com/MilesEngineering/MsgTools/pull/141
There were two bugs:
1) msgtools.js moved
2) the function called when a message is received no longer has a param named 'msg', instead it gets an event, and needs to access the message as 'event.detail.message'.

I'm confused about why new files were created in msgtools/webapp/lib that replace the old ones in MsgApp/ (some of which were prior that moved to MsgApp/js), but the old ones were never deleted.  I don't think it makes sense to keep both, I'd be much happier with just one of them.

mgazic, 2019-02-26:
Closing this one since https://github.com/MilesEngineering/MsgTools/pull/141 fixes it.  No action taken on either bundling for deployment, or removing duplication between MsgApp and msgtools/webapp/lib.  New issue created for removing duplication at https://github.com/MilesEngineering/MsgTools/issues/142

closed 2019-02-26 by mgazic

mosminerCP, 2019-02-28:
k.  Sorry been a little slow.  Unfortunately, not top of my priority list.