MilesEngineering/MsgTools #152: Variable Length Field Parsing

Aaron-Beal-CardinalPeak, 2019-07-16:
I was looking through the source on `csvToMsg()` and noticed [this line](https://github.com/MilesEngineering/MsgTools/blob/master/msgtools/lib/msgcsv.py#L122) is used to decide whether hex strings are spread into the pieces of a field were `count > 1`:
```python
if val.startswith("0x") and len(val) > 2+fieldInfo.count*int(fieldInfo.get.size):
```

Now the passed in `val` is a string. If it's in the form `"0x00112233"` and the field was 10 uint8s, then this would decide not to spread that hex value through each of the 10 uint8s. Instead it would just store the the whole `0x00112233` into the first uint8. I'm not sure why it would be done this way. Shouldn't the condition be `len(val) > 2+2*int(fieldInfo.get.size)` instead?

mgazic, 2019-07-22:
Thanks for finding that, it's definitely wonky.

This particular feature was a request from someone at Bose.  They wanted to specify one big hex string for the contents of an array, instead of comma-delimited hex strings with one value for each array element.  I think the person at Bose said they only ever wanted to specify a hex string that held exactly the right number of bytes.  So specifying 4 bytes (0x00112233) out of an array of 10 uint8s would be an error, according to what they said.

Maybe I should do the check as you suggest, and if any more than one array element's worth of bytes is specified, it should store them across the array.  If not, I probably ought to check for len(val) == 2+2*fieldInfo.count*int(fieldInfo.get.size) or len(val) == 2+2*int(fieldInfo.get.size), and explicitly raise an error for len(val) anywhere in between.

That said, there are a few unit tests for CSV conversion here:
https://github.com/MilesEngineering/MsgTools/blob/master/msgtools/lib/Test/Test.py#L89
Lines 95,96,98 look to be testing the case the code handles.  What I ought to do first is decide on behavior if there's fewer bytes in that hex string than needed for all array elements, but more bytes than needed for one array element.  Then add the appropriate test cases, and then change the code to match.

If you want to add test cases and modify the code yourself, it'd definitely be welcome!  Otherwise I'll get to it it, but not sure when (I'm on vacation in Tonga right now).

mgazic, 2019-07-22:
Whoops, I read this all wrong.  I have test cases that do contain a string with more than one array element but fewer than 'count', and the only reason they pass is because 'count' in my test data is only 3.  I'll make a new test message with a 'count' of 10, and a string with enough data for 4 bytes, just like the example you gave.

mgazic, 2019-07-22:
I updated the test case, fixed the bug as you suggested, and made a new release 0.34.7: https://pypi.org/project/msgtools/

closed 2019-07-22 by mgazic